tuya-cloudcutter/tuya-cloudcutter #765: [DUMP] [ROM] [BK7238] Generic RGBCCT Smart Light Bulb (Surplife App)

ataa, 2025-01-03:
[RGBCCT-Bulb-BK7238-Factory.zip](https://github.com/user-attachments/files/18304450/RGBCCT-Bulb-BK7238-Factory.zip)
[RGBCCT-Bulb-BK7238-Factory-rom.zip](https://github.com/user-attachments/files/18304451/RGBCCT-Bulb-BK7238-Factory-rom.zip)

WB43-M5 V1.1 Module

Cossid, 2025-01-03:
Surplife isn't a part of the Tuya platform, and this device can't be supported by CloudCutter as a result.

closed 2025-01-03 by Cossid

divadiow, 2025-08-15:
@ataa can I ask where you got your bulb and if it was Ali Express do you have the link to store/product please?

ataa, 2025-08-15:
@divadiow Got it from Amazon, they used to have BL602 chip on Module , they silently changed it to BK7238, I remember you helped me on Elektroda with some experimental firmware for it.

Every single bulb I recently came across with "Surplife" app and "10W SMART+" marking, has this chip. they sell it in two packs and 10 packs for about $4 to $4.5 each. 

https://www.amazon.ae/SKY-TOUCH-2Pcs-Smart-Led-Bulb/dp/B0BT9LPRMB

Most recent ones like these, have chip on pcb instead of module. That's even cheaper ($3.7 each)

https://www.amazon.ae/EnvoMart-WIFI-Smart-Wi-Fi-Light/dp/B0CX5CS2M5

[image]