tuya-cloudcutter/tuya-cloudcutter #766: Tuya Cloudcutter Stalls During Firmware Update and Endpoint Processing

clairekardas, 2025-01-04:
First of all thank you for the amazing work you have been doing on this project. I have been trying to get Tuya Cloudcutter to work for a few days now, testing two different Tuya generic devices. I encountered the following issues during my latest attempts at the firmware update process:

**Device 1: tuya-generic-s18-ir-repeater**
- The flash process stalls at "Firmware update progress: 91%".
- When attempting to process the endpoints, I receive a "response not found" error.

**Device 1 (Second Flash Attempt)**
- The flash stalls at:
  `[23:47:52.033949 MQTT Received] Topic: smart/device/out/zFccTvun1FjLCleDLGsG - Message: {"protocol":16,"data":{"progress":"37","firmwareType":0}}`

**Device 2: tuya-generic-p01-pir-motion-sensor-v2.1.8**
- The flash process stalls after 6 DNS queries.

**Logs:**

<details> 
  <summary>Log 1 (Device 1 - First Flash Attempt):</summary>

```
Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

[?] Select your custom firmware file for BK7231N chip: ESPHome-Kickstart-v23.08.[?] Select your custom firmware file for BK7231N chip: OpenBeken-v1.17.697_bk7231n.ug.bine-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
   ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
 ► OpenBeken-v1.17.697_bk7231n.ug.bin

Selected Device Slug: tuya-generic-s18-ir-repeater
Selected Profile: oem-bk7231n-irbox-mol-ty-2.0.0-sdk-2.3.1-40.00
Selected Firmware: OpenBeken-v1.17.697_bk7231n.ug.bin

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
.....
Found access point name: "SmartLife-88DE", trying to connect...
Device 'wlan0' successfully activated with '19886577-feec-4ee0-b1f5-993b68f49aa2'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/2LWd9nuYgV8P.deviceconfig
Saved device config in /work/configured-devices/2LWd9nuYgV8P.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
.
Found access point name: "SmartLife-88DE", trying to connect...
Error: Connection activation failed: 802.1X supplicant took too long to authenticate.
..
Found access point name: "A-88DE", trying to connect...
Device 'wlan0' successfully activated with '11d4ec2e-4dce-4130-a8e7-14cc416f3f41'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
[... 399 lines cut ...]
Jan  3 23:57:27 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.12
Jan  3 23:57:27 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1

```

</details>

Steps Taken:

- I have tried flashing both devices multiple times.
- I have tried using different device profiles for device 2 (tuya-generic-p01-pir-motion-sensor-v2.1.8, tuya-generic-pir-motion-sensor)
- I’ve attached verbose logs for two of the attempts.
- I have ensured the devices are correctly powered and connected.

Expected Behavior:

The firmware update should proceed without stalling and OpenBeken be installed and a new AP created by the tuya device.

Additional Information:

- The firmware update seems to freeze or stall at specific progress points.
- I’ve ensured that the devices are properly connected, that the network is stable and that devices are in reach.
- I am using a RaspberryPi 3B+.

Please let me know if there are any additional steps I can try or if you need further information to help resolve the issue. Thank you!

Cossid, 2025-01-04:
Define "in reach".  AP mode is not very strong on most devices, and able to connect does not mean a strong connection.  All of these logs seem to indicate weak connection, as nothing is explicitly failing, but simply timing out.  There is nothing that can change after the firmware update process has begun, there is nothing we can tweak.

clairekardas, 2025-01-04:
Thank you for the clarification on "in reach". For Device 1, your suggestion worked perfectly; moving it from 5 meters away to just a few centimeters resolved the issue.

As for Device 2, it was already positioned closer, but I’ve now adjusted it so both devices are literally right next to each other during the flashing process.
<details> 
  <summary>Here is a new log for further analysis:</summary>
Performing safety checks to make sure all required ports are available
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Safety checks complete.

[?] Select your custom firmware file for BK7231N chip: OpenBeken-v1.17.697_bk7231n.ug.bin
   ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
 ► OpenBeken-v1.17.697_bk7231n.ug.bin

Selected Device Slug: tuya-generic-p01-pir-motion-sensor-v2.1.8
Selected Profile: bk7231n-lowpower-common-ty-2.1.8-sdk-2.3.1-40.00
Selected Firmware: OpenBeken-v1.17.697_bk7231n.ug.bin

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
.
Found access point name: "A-AA7B", trying to connect...
Device 'wlan0' successfully activated with 'b48fd4cb-c5f8-4f42-8506-93baacec5db5'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/Q3aUtttj31po.deviceconfig
Saved device config in /work/configured-devices/Q3aUtttj31po.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP

Found access point name: "A-AA7B", trying to connect...
Device 'wlan0' successfully activated with 'b48fd4cb-c5f8-4f42-8506-93baacec5db5'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Flashing custom firmware...

================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
================================================================================

Using WLAN adapter: wlan0
Jan  4 01:06:31 dnsmasq[16]: started, version 2.85 cachesize 150
Jan  4 01:06:31 dnsmasq[16]: compile time options: IPv6 GNU-getopt DBus no-UBus i18n IDN2 DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC loop-detect inotify dumpfile
Jan  4 01:06:31 dnsmasq-dhcp[16]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Jan  4 01:06:31 dnsmasq-dhcp[16]: DHCP, sockets bound exclusively to interface wlan0
Jan  4 01:06:31 dnsmasq[16]: read /etc/hosts - 5 addresses
Configuration file: /dev/stdin
Using interface wlan0 with hwaddr b8:27:eb:18:f4:f6 and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
[... 81 lines cut ...]
[01:07:24.454828 LOG (Client)] ==== Request body ===
{"type":0,"upgradeStatus":2,"t":1735952843}
[01:07:24.455140 LOG (Client)] ==== End request body ===
[01:07:24.457261 LOG (Server)] Response:  {'t': 1735952844, 'success': True}
[01:07:24.683653 MQTT Received] Topic: smart/device/out/VeU7O7BRZFZUC4bGK2Ua - Message: {"protocol":4,"t":1735952844,"data":{"dps":{"1":"pir"}}}

[01:07:24.869678 Log (Client)] Request: HTTPServerRequest(protocol='http', host='10.42.42.1', method='GET', uri='/files/OpenBeken-v1.17.697_bk7231n.ug.bin', version='HTTP/1.1', remote_ip='10.42.42.12')
[01:07:24.870140 LOG (Server)] Response:  bytes=0-504271
[01:07:24.952101 MQTT Received] Topic: smart/device/out/VeU7O7BRZFZUC4bGK2Ua - Message: {"protocol":4,"t":1735952844,"data":{"dps":{"3":"low"}}}
[01:07:25.071393 MQTT Received] Topic: smart/device/out/VeU7O7BRZFZUC4bGK2Ua - Message: {"protocol":16,"data":{"progress":"0","firmwareType":0}}
[01:08:02.752895 Firmware Upload] /files/OpenBeken-v1.17.697_bk7231n.ug.bin send complete, request range: bytes=0-504271/504272
Jan  4 01:08:44 dnsmasq[16]: query[A] m2.tuyaeu.com from 10.42.42.12
[... cut ...]
I understand that there is not much to do on your end, but maybe there is something I am missing. I also want to note that I tried flashing this tuya device before using a different linux machine that had an issue when it came to creating an AP, so I switched to the pi. From what I understand, this cant be related, because with each run the flashing process starts over, and as long as I select the right device profile, it will work with the A- device from the old machine. Correct? I appreciate your help and quick response!

Cossid, 2025-01-04:
Battery devices can be tricky, they are not designed to stay connected.  Try pressing the button briefly every few seconds and see if that improves the situation, it should help prevent it from suspending into battery saving mode.

Cossid, 2025-01-09:
Closing as stale, and with the note that there is no guarantee battery devices will stay online long enough to complete an OTA, they are designed to sleep often.

closed 2025-01-09 by Cossid