tuya-cloudcutter/tuya-cloudcutter #769: IR-RF Universal Remote Controller v2.0.15 -  no 3d party firmware flash possible

nenonano, 2025-01-06:
Hi there

I've recently got a IR-RF Universal remote which looks exactly as this one: IR-RF Universal Remote Controller v2.0.8 
here's an image from Aliexpress
[image]

Smart life app recognizes it as "IR Controller Pro" and firmware is 2.0.15
[image]

tried many profiles with no success

`utente@raspberrypi:~/tuya-cloudcutter $ sudo bash tuya-cloudcutter.sh -v -r -t 30
Building cloudcutter docker image
[+] Building 1.3s (13/13) FINISHED                                                                                                                                                                                            docker:default
 => [internal] load build definition from Dockerfile                                                                                                                                                                                    0.1s
 => => transferring dockerfile: 462B                                                                                                                                                                                                    0.0s
 => [internal] load metadata for docker.io/library/python:3.9.18-slim-bullseye                                                                                                                                                          0.6s
 => [internal] load .dockerignore                                                                                                                                                                                                       0.1s
 => => transferring context: 2B                                                                                                                                                                                                         0.0s
 => [base 1/2] FROM docker.io/library/python:3.9.18-slim-bullseye@sha256:9ac27d4ecadc3ef02f980a8e2b37c7e8cdbf24039c5eddd9f98aff00d23b9e03                                                                                               0.0s
 => [internal] load build context                                                                                                                                                                                                       0.1s
 => => transferring context: 3.02kB                                                                                                                                                                                                     0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto                                                                    0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv                                                                                                                                                                           0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                                                                                                                                                                                     0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                                                                                                                                                                                0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy                                                                                                                                            0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                                                                                                                                                                              0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                                                                                                                                                                               0.0s
 => exporting to image                                                                                                                                                                                                                  0.0s
 => => exporting layers                                                                                                                                                                                                                 0.0s
 => => writing image sha256:aa0b2d9d4b0193e59f74df34b57b345a6bd6480353e784af352297351432f0c6                                                                                                                                            0.0s
 => => naming to docker.io/library/cloudcutter                                                                                                                                                                                          0.0s
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By manufacturer/device name
 ► By manufacturer/device name
   By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the brand of your device: Tuya Generic
   TEQOOZ
   TESLA
   TOMZN
   tongou
   Topgreener
   Treatlife
 ► Tuya Generic
   Ultrabrite
   UltraPro
   UME
   US Solid
   Ustellar
   Wipro

[?] Select the article number of your device: IR-RF Universal Remote Controller v2.0.8
   HWGY004 Cabinet Lamp
   IH-MC01 Window Sensor v1.0.10
   iH001 LED Controller Single Color
   IR 8 Function Fairy String Lights
   IR Remote Control with Temperature and Humidity Display
   IR-RF Universal Remote Controller v2.0.0
 ► IR-RF Universal Remote Controller v2.0.8
   KS-10WBU Brazil 2-CH Wall Socket with USB
   KWS-302WF 8-in-1 Power Meter v2.1.6
   L31 Water Leak Water Level Sensor
   L502 Water Leak Sensor v1.1.9
[... 39 lines cut ...]
output=/work/configured-devices/adnNrAUEq1c5.deviceconfig
Saved device config in /work/configured-devices/adnNrAUEq1c5.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
.
Found access point name: "SmartLife-0DF8", trying to connect...
Device 'wlan0' successfully activated with '3821a988-2cf3-45a9-a0c6-43ecd9418c4a'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================
`
I don't see any 2.0.15 firmware profiles

Is there anything I can do?

TIA

Cossid, 2025-01-07:
I was able to locate an update, and luckily it was not patched.

I have added a generic profile as `Tuya Generic IR-RF Universal Remote Controller v2.0.15` which should now work for your device.

closed 2025-01-07 by Cossid

nenonano, 2025-01-11:
Hi, worked perfectly
Thank you