tuya-cloudcutter/tuya-cloudcutter #771: Device Dump - LSC 3202088 Plug With Energy Monitoring

Betacart, 2025-01-07:
Main Module: **V1.1.17**
Other Type: **V1.0.4**

Chip Type: **BK7231N**

[image]
[ltchiptool_bk72xx_2025-01-07_14-14-53.zip](https://github.com/user-attachments/files/18333807/ltchiptool_bk72xx_2025-01-07_14-14-53.zip)

Cossid, 2025-01-07:
Plug firmware 1.1.17 is already on the [known patched firmware](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware) list, serial flashing will remain the only option for this device.

closed 2025-01-07 by Cossid

Cossid, 2025-01-07:
Use the `Analyze Firmware Dump` option instead, and choose your dump file.

Betacart, 2025-01-07:
This gives me following result:
`W: UPK: Skipping LED 1 on pin P6 - shared status LED is enabled
I: UPK: Found BK7231N config!
I: UPK: Switch/plug config
I: UPK:  - relay 1: pin P8
I: UPK:  - LED 1: (shared)
I: UPK:  - button 1: pin P7
I: UPK: Status LED: pin P10 (shared), inverted False
I: UPK: Power monitoring chip BL0937: CF/ELE=P26, CF1/VI=P24, SEL=P11
I: UPK:  - shunt resistor: 1 mΩ
`