tuya-cloudcutter/tuya-cloudcutter #773: Device Dump: DY-RQ400A Tuya Combustible Gas Alarm

lanrat, 2025-01-09:
[DY-RQ400A.zip](https://github.com/user-attachments/files/18355763/DY-RQ400A.zip)

chip info:
```
I: +-----------------------+-------------------------------------+
I: | Name                  | Value                               |
I: +-----------------------+-------------------------------------+
I: | Protocol Type         | FULL                                |
I: | Chip Type             | BK7231N                             |
I: | Bootloader Type       | BK7231N 1.0.1                       |
I: | Chip ID               | 0x7231c                             |
I: | Boot Version String   | N/A                                 |
I: |                       |                                     |
I: | MAC Address           | 38:A5:C9:42:EF:92                   |
I: |                       |                                     |
I: | Flash ID              | EB 60 15                            |
I: | Flash Size (by ID)    | 2 MiB                               |
I: | Flash Size (detected) | 2 MiB                               |
I: |                       |                                     |
I: | Encryption Key        | 510fb093 a3cbeadc 5993a17e c7adeb03 |
I: +-----------------------+-------------------------------------+
I: |-- Finished in 11.965 s
```

Note: This DY-RQ400A contains a CB2S Wi-Fi Module with a BK7231N. There is another version of the DY-RQ400A with the WB2S/BK7231T.

lanrat, 2025-01-09:
ltchiptool/UPK2ESPHome is uanble to parse the firmware dump that is generated. Multiple dumps were taken, all with the same hash.

[image]

Cossid, 2025-01-09:
This is firmware 3.1.17 which is already on the known patched firmware list.  Currently, serial flashing is the only solution for this device, but it is a version of TuyaOS that will have a new OTA exploit in the future (no set date yet).

This is a TuyaMCU device, and there is no pin configuration for it, you'll need to map out DPIDs over tuya's uart communication.

Here are the DPIDs for this device:

```json
"model": {
    "modelId": "exos58",
    "services": [{
            "actions": [],
            "code": "",
            "description": "",
            "events": [],
            "name": "\u9ed8\u8ba4\u670d\u52a1",
            "properties": [{
                    "abilityId": 1,
                    "accessMode": "ro",
                    "code": "checking_result",
                    "description": "",
                    "extensions": {
                        "iconName": "icon-tcl_function_vs2",
                        "trigger": "direct"
                    },
                    "name": "\u8bbe\u5907\u81ea\u68c0\u7ed3\u679c",
                    "typeSpec": {
                        "type": "enum",
                        "range": [
                            "checking",
                            "check_success",
                            "check_failure",
                            "others"
                        ]
                    }
                }, {
                    "abilityId": 2,
                    "accessMode": "ro",
                    "code": "gas_sensor_state",
                    "description": "",
                    "extensions": {
                        "iconName": "icon-huoyan",
                        "trigger": "direct"
                    },
                    "name": "\u71c3\u6c14\u68c0\u6d4b\u72b6\u6001",
                    "typeSpec": {
                        "type": "enum",
                        "range": [
                            "alarm",
                            "normal"
                        ]
                    }
                }, {
                    "abilityId": 3,
                    "accessMode": "rw",
                    "code": "alarm_time",
                    "description": "",
                    "extensions": {
                        "iconName": "icon-timer1"
                    },
                    "name": "\u62a5\u8b66\u65f6\u957f",
                    "typeSpec": {
                        "type": "value",
                        "max": 180,
                        "min": 1,
                        "scale": 0,
                        "step": 1,
                        "unit": "s"
                    }
                }, {
                    "abilityId": 5,
                    "accessMode": "ro",
                    "code": "gas_sensor_value",
                    "description": "",
                    "extensions": {
                        "iconName": "icon-jiare1",
                        "trigger": "direct"
                    },
                    "name": "\u71c3\u6c14\u68c0\u6d4b\u503c",
                    "typeSpec": {
                        "type": "value",
                        "max": 1000,
                        "min": 0,
                        "scale": 1,
                        "step": 1,
                        "unit": "LEL"
                    }
                }, {
                    "abilityId": 8,
                    "accessMode": "rw",
                    "code": "alarm_ringtone",
                    "description": "",
                    "extensions": {
                        "iconName": "icon-yinliang"
                    },
                    "name": "\u62a5\u8b66\u94c3\u58f0",
                    "typeSpec": {
                        "type": "enum",
                        "range": [
                            "1",
                            "2",
                            "3",
                            "4",
                            "5"
                        ]
                    }
                }, {
                    "abilityId": 9,
                    "accessMode": "rw",
                    "code": "self_checking",
                    "description": "",
                    "extensions": {
                        "iconName": "icon-yunhang"
                    },
                    "name": "\u8bbe\u5907\u81ea\u68c0",
                    "typeSpec": {
                        "type": "bool"
                    }
                }, {
                    "abilityId": 10,
                    "accessMode": "ro",
                    "code": "preheat",
                    "description": "",
                    "extensions": {
                        "iconName": "icon-dp_light2"
                    },
                    "name": "\u9884\u70ed",
                    "typeSpec": {
                        "type": "bool"
                    }
                }, {
                    "abilityId": 12,
                    "accessMode": "rw",
                    "code": "alarm_switch",
                    "description": "",
                    "extensions": {
                        "iconName": "icon-power"
                    },
                    "name": "\u62a5\u8b66\u5f00\u5173",
                    "typeSpec": {
                        "type": "bool"
                    }
                }, {
                    "abilityId": 14,
                    "accessMode": "rw",
                    "code": "muffling",
                    "description": "",
                    "extensions": {
                        "iconName": "icon-Mute"
                    },
                    "name": "\u6d88\u97f3",
                    "typeSpec": {
                        "type": "bool"
                    }
                }
            ]
        }
    ]
}
```

closed 2025-01-09 by Cossid

lanrat, 2025-01-09:
> This is a TuyaMCU device, and there is no pin configuration for it, you'll need to map out DPIDs over tuya's uart communication.

Is there any documentation for DPIDs so that I can generate my own ESPHome based firmware for this device and flash it over serial?

Thanks.

lanrat, 2025-01-10:
Where did the json DPIDs you posted come from? Was it generated from the dumps I provided?

lanrat, 2025-01-11:
The WB2S/BK7231T version of the DY-RQ400A has already been analysed here: https://devices.esphome.io/devices/Tuya-DY-RQ400A-Combustible-Gas-Alarm

The DPID's you posted above seem to match the WB2S/BK7231T so I flashed a ESPHome config to mine over serial with the TUYA config from the other version. The alarm test seems to work, but nothing else. ESPHome also constantly complains that the Tuya component failed to initialize with the state 0.

Is this a result of the TuyaOS 3 communicating differently over UART?

Would there be any work I can do to help further support for TuyaOS 3?

Cossid, 2025-01-11:
No, the UART protocol wouldn't have changed with TuyaOS 3, as the protocol is also in the TuyaMCU firmware, and they'd have to match.  That protocol hasn't changed as far as we're aware.

They are attached to the Product Id in the TuyaMCU firmware.

It is possible this device is possibly effected by the buffer size issue that is currently being worked on at the following locations:
https://github.com/esphome/issues/issues/6240
https://github.com/libretiny-eu/libretiny/pull/313

So it may be worth trying out those changes.

Otherwise it is possible that while the DPIDs match, the TuyaMCU firmware isn't exactly the same.  The device you supplied has a product id of `stqmzk01tbm3qwhg` while the one in that devices page has an id of `13nrj1aeeaqh54cz`

lanrat, 2025-01-11:
The Tuya MCU on mine `stqmzk01tbm3qwhg` is different than the `13nrj1aeeaqh54cz`. 

The  `13nrj1aeeaqh54cz` has a `RJM8L151F6P6`, while the `stqmzk01tbm3qwhg` has a `003FQ6` which I'm unable to find any data on.

[image]

lanrat, 2025-01-11:
Also very odd, I just tested the device again with the same ESPHome firmware that I left on it with the tuya init_state 0 errors, and almost every function worked. A few of the toggles seem to get inverted if you change them too fast, but it _seems_ to work for the most part. Nothing changed since I tried a few days ago with the errors....

I'll try adding the buffer fix and see if that helps.

Cossid, 2025-01-11:
The chip itself doesn't really matter, it's the firmware that is on it.  That product id is programmed into the firmware of the TuyaMCU.

Yeah, that sounds like it could be running into the buffer issue then, where the message just gets truncated, and nothing registers, but it might recover with smaller messages over time.

lanrat, 2025-01-12:
With the serial buffer fix it seems to be much more stable. I'll keep it running for a few days and report back if anything changes.

Thanks for your help.

Cossid, 2025-01-12:
I have a similar device (also a natural gas sensor, but yet a different model), and it also would reboot every couple minutes, but randomly stay on for a day.

I am also giving the serial buffer patch a try, and so far, it seems to also be going well.

divadiow, 2025-01-12:
kind of interesting is that the N version has a TuyaMCU OTA fw upgrade if not already on version 1.0.1

[image]
[Tuya_DY-RQ400A_MCU_1.0.1_003FQ6.zip](https://github.com/user-attachments/files/18388392/Tuya_DY-RQ400A_MCU_1.0.1_003FQ6.zip)