tuya-cloudcutter/tuya-cloudcutter #774: No items in custom profiles

Sazerak, 2025-01-11:
Hi, I'm trying to flash my device using a custom profile, but when I go through the interactive prompts and choose "From device-profiles", there is nothing listed. At first I thought it was because I hadn't followed the proper naming convention, but even after renaming, nothing shows up. Not even the provided profiles show up in the list (there are 4 profiles in the custom-firmware directory by default). Any tips?

Cossid, 2025-01-11:
It sounds like you are mixing up device profiles and custom firmware, they are two different things.
A device profile constitutes at least a directory in the `device-profiles` directory and within must be at least a `device.json` and `profile.json` file.

Custom firmware is limited to names that contain the platform in them, in order to prevent mis-flashing, and is narrowed down by the selection device profile selected.

You'll need to provide more context with sample files for more support.

Cossid, 2025-01-11:
I should add, you may be unclear, you select your device first, then your firmware, `from device-profiles` doesn't mean custom firmware, that part comes later after you select your device.

Sazerak, 2025-01-11:
Ah yes, you're right, I am getting the two confused. I just looked again in the [Flashing Custom Firmware](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/INSTRUCTIONS.md#flashing-custom-firmware) instructions and it says I need a Profile AND Firmware. I thought it was one or the other.

So obviously I'm missing a step in my knowledge. Basically, I'm trying to flash an LED light like the one [HERE](https://www.elektroda.com/rtvforum/topic4024658.html) only I'm trying (hoping) to flash ESPHome instead of OpenBeken (I'm not familiar at all with OpenBeken). I've created my YAML file on ESPHome and exported both the UF2 and UG files (tried both just in case). I thought it was as simple as adding these into the custom-firmware folder and following the prompts, but I'm starting to realize it's not.

What would be the best way to proceed from here? I was also considering trying out the provided ESPHome Kickstart file from the custom-firmwares to keep me a starting point, but I guess I would need a profile for this too.

Originally, I planned on going the route of choosing by firmware version, but that one profile multiple profiles, none of which match my device, so I wasn't sure if I could actually try any of them.

Cossid, 2025-01-11:
If there isn't already a profile for your specific device, https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#how-do-i-find-out-what-firmware-version-my-device-has

But also be aware of the [known patched firmware list](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware), part of the reason why there might not be a profile is because it is patched.

Cossid, 2025-01-11:
Based on the discussion in that linked thread, if you have firmware version 1.0.5, that is unfortunately patched and will not work with CloudCutter.

Sazerak, 2025-01-11:
Ah, sonavagun! I didn't realize there was a known patched list. Sure enough, my light is on there. Oh well, I guess I'm doing this the old-fashioned way then. At least having to do it that way, I'll be able to check the PWM mapping while I'm there since the PWM mapping from OpenBeken isn't something I'm familiar with.

closed 2025-01-11 by Sazerak

Cossid, 2025-01-11:
Unfortunately, the configuration for this device isn't in storage, but rather firmware, so the analysis tools will fail to generate a config.

Instead, the config for this one is stored within the firmware itself, but it has been extracted.

You can paste;

```json
{
	"0err": 70,
	"1err": 40,
	"Jsonver": "1.0.2",
	"b_lv": 1,
	"b_pin": 6,
	"bitseq": 0,
	"brightmax": 100,
	"brightmin": 10,
	"briseg": 10,
	"buzzer": 0,
	"c_lv": 1,
	"c_pin": 8,
	"cagt": 20,
	"category": "0505",
	"cctseg": 10,
	"cmod": "rgbcw",
	"colormax": 100,
	"colormaxp": 100,
	"colormin": 10,
	"colorpfun": 1,
	"crc": 64,
	"customcode": 32382,
	"cwmaxp": 100,
	"cwtype": 1,
	"defbright": 100,
	"defcolor": "c",
	"deftemp": 0,
	"dmod": 0,
	"g_lv": 1,
	"g_pin": 7,
	"gmkb": 60,
	"gmkg": 60,
	"gmkr": 80,
	"gmwb": 75,
	"gmwg": 70,
	"gmwr": 100,
	"ir": 23,
	"irfunc": 1,
	"irk10fun": 11,
	"irk10val": 54,
	"irk11fun": 30,
	"irk11val": 24,
	"irk12fun": 33,
	"irk12val": 55,
	"irk1fun": 1,
	"irk1val": 5,
	"irk2fun": 50,
	"irk2val": 50,
	"irk3fun": 51,
	"irk3val": 26,
	"irk4fun": 52,
	"irk4val": 51,
	"irk5fun": 53,
	"irk5val": 27,
	"irk6fun": 20,
	"irk6val": 48,
	"irk7fun": 39,
	"irk7val": 25,
	"irk8fun": 40,
	"irk8val": 53,
	"irk9fun": 10,
	"irk9val": 22,
	"keynumber": 12,
	"leaderr": 15,
	"module": "CBU",
	"mutex": 1,
	"nightled": 0,
	"notdisturb": 0,
	"onoffmode": 0,
	"onofftime": 800,
	"pairt": 18,
	"pmemory": 1,
	"prodagain": 0,
	"pwmhz": 1000,
	"r_lv": 1,
	"r_pin": 26,
	"remdmode": 1,
	"rgbt": 10,
	"rstbr": 50,
	"rstcor": "c",
	"rstnum": 3,
	"rsttemp": 0,
	"starterr": 70,
	"title20": 1,
	"w_lv": 1,
	"w_pin": 24,
	"wfcfg": "spcl",
	"wfct": 3,
	"whiteseg": 10,
	"wt": 20
}
```
into the input data section of https://upk.libretiny.eu/ and it will generate a basic ESPHome config for you.

Sazerak, 2025-01-11:
OK, now you've really intrigued me. I spent a bunch of time last night trying to map the PWMs since the mapping from that post I linked made no sense to me, but the resulting YAML from what you posted seems like it should work. Where did you find this and are you sure it's for my exact device? While it looks like it would work, the only part that really puzzles me is that it uses RGBCW (at least according to the resulting YAML) whereas the device should be a RGBWW. That being said, I think RGBWW controls warm and cold white separately which doesn't seem to be the case for this device, so maybe it's not actually a RGBWW.

I'm not overly familiar with RGB lighting, this is the first one I've tried to figure out on my own, so I'm learning a lot!

Cossid, 2025-01-11:
Yes, it is the same device, it is actually from that same thread, provided to me elsewhere, but that is how it got added to the known patched list.

RGBWW vs RGBCT is just how the white channels are handled, but both are still PWM in this case.
WW just has a channel for warm white and a channel for cold white, and the brightness is handled via PWM.
CT has a channel for Color Temperature and a channel for brightness.  Brightness handles brightness for the white channel as a whole, and Color Temperature will shift the white color temperature.

(and you may need to invert P24 to get the temperature to match, the converter doesn't understand that).

This device is indeed RGBCT

Sazerak, 2025-01-11:
Thanks for the info! Yes I had seen about P24 being possibly inverted. I'll try it like that first and see what happens. I definitely agree that this light is probably a CT instead of WW (especially considering the comment in that thread I posted earlier about inverting P24).

I should be good from here, thanks for all your help!