tuya-cloudcutter/tuya-cloudcutter #775: Custom firmware not detected

slawrensen, 2025-01-15:
For some reason, the custom firmware in the custom firmware folder never shows up in the list. I read the readme in the custom firmware folder and the files are named star_nebula_bk7231t.ota.ug.bin and star_nebula_bk7231n.ota.ug.bin, selecting either version of the SK20.

Cossid, 2025-01-15:
That means the files are not actually valid.  They must contain the correct magic bytes within the file, to make sure it is the platform and type it is expected to be.

For example, you can't rename `star_nebula_bk7231t.uf2` to `star_nebula_bk7231t.ota.ug.bin` because that's not what it is.  There is a protection both built in for each type of file, and another for platform verification to match the platform of the profile you've selected.

Without more context, there isn't any additional support that can be provided.

slawrensen, 2025-01-15:
Does the below context add more clarity? I'm compiling with esphome and then copying the file over with its original name (image_bk7231n_app.ota.ug.bin). I'm clearly doing something wrong, so I appreciate any help including a point in the right direction.

platform_bk72xx.yaml
```
---
bk72xx:
  board: cbu
```

recommended_base.yaml

```
substitutions:
  device_name: "Star Projector"

esphome:
  name: "nebula"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none

  ap:
    ssid: "Fallback AP"
    password: !secret wifi_fallback_password

captive_portal:

logger:
  level: INFO

ota:
  platform: esphome
  password: !secret ota_password

web_server:
  port: 80

# by default controlling is done via the button and via the web interface
# if you use homeassistant you can uncomment below code:
api:
  password: !secret api_password

# or via MQTT:
#mqtt:
#  broker: !secret mqtt_broker

packages:
  local_package_1: !include platform_bk72xx.yaml
  local_package_2: !include nebula_light_device.yaml
```
command line output:
```

(venv) slawrensen@thinkpad:~/sk20$ esphome compile recommended_base.yaml 
INFO ESPHome 2024.12.2
INFO Reading configuration recommended_base.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing nebula (board: cbu; framework: arduino; platform: libretiny @ 1.7.0)
-------------------------------------------------------------------------------------
Warning! Non-Git installations are NOT SUPPORTED.
HARDWARE: BK7231N 120MHz, 256KB RAM, 1.03MB Flash
 - framework-arduino-api @ 2022.8.24+sha.237b10a 
 - framework-beken-bdk @ 0.0.0+v2021.06.07.sha.6491b8c 
 - library-flashdb @ 1.2.0+sha.d5c892f 
 - library-freertos @ 9.0.0+sha.95cc959 
 - library-freertos-port @ 2023.5.23+sha.a917d93 
 - library-lwip @ 2.2.0-bdk+sha.48da4a2 
[... 658 lines cut ...]
Compiling .pioenvs/nebula/bdk_core/driver/entry/boot_handlers.S.o
Compiling .pioenvs/nebula/bdk_core/driver/entry/boot_vectors.S.o
Compiling .pioenvs/nebula/bdk_core/func/wlan_ui/wlan_ui.c.o
Archiving .pioenvs/nebula/libbdk_app.a
Indexing .pioenvs/nebula/libbdk_app.a
Archiving .pioenvs/nebula/libbdk_driver.a
Indexing .pioenvs/nebula/libbdk_driver.a
Archiving .pioenvs/nebula/libbdk_core.a
Indexing .pioenvs/nebula/libbdk_core.a
Linking .pioenvs/nebula/raw_firmware.elf
|-- Image 1: raw_firmware.elf
|   |-- image_bk7231n_app.0x011000.rbl
|   |   |-- raw_firmware.bin
|   |-- image_bk7231n_app.0x011000.crc
|   |-- image_bk7231n_app.0x129F0A.rblh
|   |-- image_bk7231n_app.ota.rbl
|   |-- image_bk7231n_app.ota.ug.bin
RAM:   [====      ]  36.7% (used 96092 bytes from 262144 bytes)
Flash: [========  ]  84.3% (used 913408 bytes from 1083136 bytes)
Building UF2 OTA image
|-- esphome_2024.12.2_cbu_bk7231n_lt1.7.0.uf2
|-- firmware.uf2
|-- firmware.bin

```

Cossid, 2025-01-15:
There is no reason copying the generated `.ota.ug.bin` file should not work, but you can also copy the `.uf2` file as well, either are supported.  They must be copied into the `custom-firmware` directory.  There isn't much more to it than that.

The prompt to select custom firmware is automatically filtered down by the platform of the profile you've chosen (IE, all BK7231T options or all BK7231N options).

Alternately, you can just flash the kickstart bins that are included, connect to the AP, and upload the firmware through the web interface.

slawrensen, 2025-01-15:
Nvm, I got it sorted after running through the procedure again. I can confirm the change below is the only one needed for version 2.0.2 of the SK20. Initially, getting it into Wi-Fi mode was a bit tricky. Entering fast flash mode is straightforward—just hold the button for 7 seconds (repeat if necessary). To switch to slow flash mode, unplug and replug the power adapter three times.

platform_bk72xx.yaml
```

---
bk72xx:
  board: cbu

```

closed 2025-01-15 by slawrensen