tuya-cloudcutter/tuya-cloudcutter #777: TH01 Temp Humidity Sensor. Stuck in flashing Loop - !!! Endpoint response not found, using default response - tuya.device.dev.dp.get

flewis88, 2025-01-21:
Tuya Generic TH01 Temp & Humidity Sensor reports:
Main Module v2.1.8
MCU Module v1.0.0

Using the v2.1.8 common fly firmware, the flashing process ends up in a loop, repeating the "Endpoint response not found" as below. Logs below are looping over approx 4 hours, without completion or termination.

`Using WLAN adapter: wlan0
Configuration file: /dev/stdin
Using interface wlan0 with hwaddr b8:27:eb:15:39:25 and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
If your device gets stuck here with no progress after several (at least two) minutes, see https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#my-device-gets-stuck-after-dhcp-what-can-i-do for additional steps
Using PSK v1 - Received PSK ID version 01
Processing endpoint /v2/url_config
Processing endpoint tuya.device.active
Processing endpoint tuya.device.dynamic.config.get
Processing endpoint tuya.device.upgrade.get
Processing endpoint tuya.device.upgrade.status.update
Processing endpoint tuya.device.dev.dp.get
!!! Endpoint response not found, using default response - tuya.device.dev.dp.get (This is usually okay and safe to ignore unless something isn't working)
Processing endpoint /files/ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
error on read: [Errno 113] No route to host
[Firmware Upload] /files/ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin send complete, request range: bytes=0-561887/561888
Using PSK v1 - Received PSK ID version 01
Processing endpoint /v2/url_config
Processing endpoint tuya.device.dynamic.config.get
Processing endpoint tuya.device.dev.dp.get
!!! Endpoint response not found, using default response - tuya.device.dev.dp.get (This is usually okay and safe to ignore unless something isn't working)
Using PSK v1 - Received PSK ID version 01
Processing endpoint /v2/url_config
Processing endpoint tuya.device.dynamic.config.get
Processing endpoint tuya.device.dev.dp.get
!!! Endpoint response not found, using default response - tuya.device.dev.dp.get (This is usually okay and safe to ignore unless something isn't working)
Using PSK v1 - Received PSK ID version 01
Processing endpoint /v2/url_config
Processing endpoint tuya.device.dynamic.config.get
Processing endpoint tuya.device.dev.dp.get
!!! Endpoint response not found, using default response - tuya.device.dev.dp.get (This is usually okay and safe to ignore unless something isn't working)
Using PSK v1 - Received PSK ID version 01
Processing endpoint /v2/url_config
Processing endpoint tuya.device.dynamic.config.get
Processing endpoint tuya.device.dev.dp.get
!!! Endpoint response not found, using default response - tuya.device.dev.dp.get (This is usually okay and safe to ignore unless something isn't working)
Processing endpoint tuya.device.uuid.pskkey.get
Using PSK v1 - Received PSK ID version 01
Processing endpoint /v2/url_config
Processing endpoint tuya.device.dynamic.config.get
Processing endpoint tuya.device.dev.dp.get
!!! Endpoint response not found, using default response - tuya.device.dev.dp.get (This is usually okay and safe to ignore unless something isn't working)
Using PSK v1 - Received PSK ID version 01
Processing endpoint /v2/url_config
Processing endpoint tuya.device.dynamic.config.get
Processing endpoint tuya.device.dev.dp.get
!!! Endpoint response not found, using default response - tuya.device.dev.dp.get (This is usually okay and safe to ignore unless something isn't working)
Using PSK v1 - Received PSK ID version 01
Processing endpoint /v2/url_config
Processing endpoint tuya.device.dynamic.config.get
Processing endpoint tuya.device.dev.dp.get
!!! Endpoint response not found, using default response - tuya.device.dev.dp.get (This is usually okay and safe to ignore unless something isn't working)`

Cossid, 2025-01-21:
It sounds like this is probably a battery device.  It is putting itself to sleep and re-waking, or rebooting.  You may need to repeatedly press a button in order to prevent it from sleeping.

Battery devices are often not well supported, as we can't control keeping them awake, especially on the versions with a TuyaMCU like this one, as the TuyaMCU can control the power to the wifi module.

flewis88, 2025-01-21:
> It sounds like this is probably a battery device.  It is putting itself to sleep and re-waking, or rebooting.  You may need to repeatedly press a button in order to prevent it from sleeping.
> 
> Battery devices are often not well supported, as we can't control keeping them awake, especially on the versions with a TuyaMCU like this one, as the TuyaMCU can control the power to the wifi module.
> 
> `/v2/url_config` is the first thing a device calls on a new boot after network activation.  Handling the extra endpoint isn't the issue, it's the rebooting that causes it to get stuck.  CloudCutter only works during a boot when activation happened, so once it reboots, it will almost certainly never finish.

Thanks! I reckon you're on the money there, and yes it is battery powered. Will try the reset. Otherwise, do you have a reliable path to get local IPs or Mac addresses for Tuya devices? These won't show up in my DHCP lease list on my router or wifi controller, so struggling to use the basic localtuya option as it doesn't auto discover and wants an IP

Cossid, 2025-01-21:
I doubt LocalTuya works correctly with any battery powered devices, as they don't stay connected to LAN due to the sleep nature of them, and LocalTuya uses the LAN protocol.

They should still be requesting DHCP information though, but possibly only on first powered boot, as they probably do cache things for sleep/wake operation.

https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out won't get you the exact MAC address, as they shift for AP mode and such, but it will get you close to help identify.

fcol95, 2025-03-18:
Hey @flewis88!
Got the same TH01 device and similar behaviour as you got there, have you figured out how to use it with esphome after all?
Cheers!

flewis88, 2025-03-18:
> Hey @flewis88!
> Got the same TH01 device and similar behaviour as you got there, have you figured out how to use it with esphome after all?
> Cheers!

I didn't, but I think I'll need to just get a BDM and directly flash the device unfortunately! Seems the issue is being battery based and they go to sleep. Device might be unviable on ESPhome once the sleep mode is gone

fcol95, 2025-03-22:
>I didn't, but I think I'll need to just get a BDM and directly flash the device unfortunately! Seems the issue is being battery based and they go to sleep. Device might be unviable on ESPhome once the sleep mode is gone

Yes ended up doing the flash manually with a USB serial adapter, the TuyaMCU indeed brings the module to sleep periodically so had to continuously click the button on it during the flash to keep it alive long enough for the flash duration!

flewis88, 2025-03-22:
> Yes ended up doing the flash manually with a USB serial adapter, the TuyaMCU indeed brings the module to sleep periodically so had to continuously click the button on it during the flash to keep it [...]

What sort of serial adaptor did you use? Did you have to solder on wires? Is it working well on ESPhome?

fcol95, 2025-03-25:
> What sort of serial adaptor did you use? Did you have to solder on wires?
CH340C-based USB to TTL adapter.
Also, I just figured out that if you want to spare yourself from pinging continuously the reset button to keep the TuyaMCU from resetting the CB3S wifi module, you can also just short to ground pin 9 of the module! (i.e. its ground pin as seen [here](https://docs.libretiny.eu/boards/cb3s/?h=cb3s#usage).
Seems that's how the TuyaMCU kills the wifi chip to go to sleep (and not through its CEN/reset pin as i thought initially).
So this might also fix the issue with cloudcutter crashing during its process, because it looses its connectivity to the CB3S module which gets powered off by tuyaMCU (haven't tried, as i erased the original firmware on the CB3S).

> Is it working well on ESPhome?
Esphome seems to boot up fine once kept alive with grounding CB3S module pin 9 (and periodically when not force grounding the pin 9).
However, it doesn't seem to get any data from the tuyaMCU (temp, humidity, etc.).

I tried OpenBeken with [this config](https://www.elektroda.com/rtvforum/viewtopic.php?p=21442008#21442008), which now seems to work (I see the temp/humidity data coming in from time to time in my MQTT broker once configured to connect on my wifi and server).
The difference seems to be that OpenBeken has a "tmSensor" driver for battery powered devices, which from what i read sends data request commands to the tuyaMCU at wake up, instead of waiting for receiving data as ESPHome seems to do with the basic tuyaMCU driver.
Might be a workaround i don't know about to make it work in esphome, but haven't took the time too investigate yet! (might also keep it as is for the moment as it with with OBK)

Electrotechnique, 2025-05-21:
I have some Tuya TH-01 sensors with the same V2.1.8 / V1.0.0 and am using the TuyaMCU V2.1.8 profile, but cannot get the exploit to work at all. It applies it and then after a reboot of the sensor, it reports the exploit didn't work. I tried it several times. They have CB3S modules in them and I'm trying to get an ESP-12F working in their place.