tuya-cloudcutter/tuya-cloudcutter #779: Device Dump - Costco Feit PLUG3/WIFI/WP/2

AndrewKCST, 2025-01-26:
Costco has these on clearance, $8 for a 2 pack. Not bad to disassemble and serial flash, but OTA would be easier 😊. Thanks for all you do!

Cossid, 2025-01-26:
You've posted the rom backup, not the flash.  We need the full 2MiB flash to create a profile.

AndrewKCST, 2025-01-26:
@Cossid sorry. I realized that this morning. I was up way too late tinkering.

[Costco_Feit_PLUG3_WIFI_WP_2_ltchiptool_bk72xx_2025-01-26_08-52-27.zip](https://github.com/user-attachments/files/18550511/Costco_Feit_PLUG3_WIFI_WP_2_ltchiptool_bk72xx_2025-01-26_08-52-27.zip)

[image]

Cossid, 2025-01-26:
Unfortunately, 1.1.17 plug firmware is on the [known patched list](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware), serial flashing will remain the only option for this device.

closed 2025-01-26 by Cossid