tuya-cloudcutter/tuya-cloudcutter #780: Device Dump - Amazon Basics B22D smart rgb 12w bulb

pokewizardSAM, 2025-01-29:
Hey there , i recently got a cheap amazon basics 12w rgb bulb , has beken 7321N i suppose and well it runs with smartlife. I tried dumping the fw using itchiptool and when i tried to analyze the fw dump but got erros such as this 

[image]

i know it's been patched for cloud cutter as it is running v1.5.40. what i would like to know is whether this is hackable ? It would be very helpfull if you guys can add a profile for this or help me make one.
DumpFile (contains two dumps just for safety): https://fileditchfiles.me/file.php?f=/s21/tKvUMfUejkFjEitsKNMC.rar

ThankYou

Cossid, 2025-01-29:
Is the device itself actually branded Amazon Basics with an fcc id and all (Note true Amazon basics devices have a lot of regulation info printed directly on them)?  I'm wondering if this is just a knock-off product.  Do you have a link to the product you purchased?

I can see that this device has an `Amazon basic Smart bulb 5Ch RGBCCT` but I don't think that matches official Amazon Basics branding (Note `Amazon basic` vs `Amazon basics`), and all Amazon Basics devices so far have used ESP32 or ESP32-C3 chips specially made for amazon that only communicate with their own AWS IoT platform, not Tuya.

The dump you have provided is indeed for BK7231N, and is indeed patched from the CloudCutter exploit.  Since it is BK7231N and you can attach serial, you can serial flash a 3rd party firmware and it should work fine.

Cossid, 2025-01-29:
```json
{
	"Jsonver": "1.0.0",
	"b_lv": 1,
	"b_pin": 8,
	"bleonoff": 1,
	"brightmax": 100,
	"brightmin": 10,
	"c_lv": 1,
	"c_pin": 7,
	"cagt": 20,
	"category": "0505",
	"cmod": "rgbcw",
	"colormax": 100,
	"colormin": 10,
	"colorpfun": 0,
	"crc": 38,
	"cwmaxp": 100,
	"cwtype": 1,
	"defbright": 100,
	"defcolor": "c",
	"deftemp": 100,
	"dmod": 0,
	"g_lv": 1,
	"g_pin": 26,
	"gmkb": 60,
	"gmkg": 60,
	"gmkr": 80,
	"gmwb": 75,
	"gmwg": 70,
	"gmwr": 100,
	"mixway": 0,
	"module": "CB3L",
	"notdisturb": 1,
	"onoffmode": 0,
	"pmemory": 1,
	"prodagain": 0,
	"pwmhz": 1000,
	"r_lv": 1,
	"r_pin": 24,
	"remdmode": 0,
	"rgbt": 10,
	"rstbr": 50,
	"rstcor": "c",
	"rstnum": 5,
	"rsttemp": 100,
	"title20": 0,
	"w_lv": 1,
	"w_pin": 6,
	"wfcfg": "spcl_auto",
	"wfct": 3,
	"wt": 20
}
```

is the device configuration in storage, you should be able to paste that into https://upk.libretiny.eu/ to generate a config.

pokewizardSAM, 2025-01-30:
Thanks, for the help @Cossid regarding the product here is the link, and this is a real product https://www.amazon.in/Amazon-Basics-Assistant-Enabled-Multicolour/dp/B0BBMRZW5N?source=ps-sl-shoppingads-lpcontext&ref_=fplfs&smid=A2AL6IVND0I91F&th=1

Adding some photos too:

[image]
[image]
[image]
[image]
[image]
[image]

Cossid, 2025-01-30:
Interesting, thanks for the images/confirmation.

But there isn't anything further CloudCutter can do, as the firmware is known patched, so closing this issue.

closed 2025-01-30 by Cossid