tuya-cloudcutter/tuya-cloudcutter #787: aldi-casalux-wifi-led-rgb-light-strip.json 2024 different?

matzzze, 2025-02-16:
Hi,

recently bought this strip in 2025. However it doesn't work:
The initial upload of the exploit supposedly works but after powercycling the device, bringing it back to slow blinking mode and then reconnecting it complains that it wasn't successful.

Opening the device I still see its the bk7231n chip inside - anyone aware whether they changed the sw/firmware on it so that the exploits don't work anymore

Cossid, 2025-02-16:
What firmware version does yours have?
https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#how-do-i-find-out-what-firmware-version-my-device-has

matzzze, 2025-02-16:
Thanks, is v2.0.28 for both Main and MCU according to the app

Cossid, 2025-02-16:
2.0.28 is on the [known patched firmware list](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware), serial flashing will be the only option for this version.

closed 2025-02-16 by Cossid