tuya-cloudcutter/tuya-cloudcutter #788: BSEED and Bingoelec WB3S 1 gang switch

Malaysk, 2025-02-22:
I need help adding tuya-cloudcutter 1 gang switch from BSEED and Bingoelec to the database, I have about 12 of these switches.
They all have firmware 1.0.4. I tried to reflash them using tuya-cloudcutter by selecting all firmware from version 1.0.4, but they all failed (there is no firmware version 1.0.4 for switches in the database). I also tried to reflash using BK7231Flasher, but I have little knowledge and I think I soldered the wires to the board incorrectly and I also failed.
There is exactly such switch, but with firmware 1.0.7 and I managed to flash it with the command sudo ./tuya-cloudcutter.sh -p deta-6911ha-switch -f OpenBeken-v1.18.31_bk7231t.ug.bin and it works perfectly.
Here are the data of this switch from platform.tuya.com:
`
<html>
<body>

switch_1 | Boolean | "{true,false}"
-- | -- | --
countdown_1 | Integer | {   "unit": "s",   "min": 0,   "max": 86400,   "scale": 0,   "step": 1 }
relay_status | Enum | {   "range": [     "power_off",     "power_on",     "last"   ] }
light_mode | Enum | {   "range": [     "relay",     "pos",     "none"   ] }
switch_backlight | Boolean | "{true,false}"
cycle_time | String | {}
random_time | String | {}
switch_inching | String | {}

</body>
</html>
`
and more information
`
{
  "result": [
    {
      "active_time": 1669532029,
      "bind_space_id": "35651116",
      "category": "kg",
      "create_time": 1636538890,
      "custom_name": "111111",
      "icon": "smart/icon/001461896993159bfIWI/7fd9abcedb4b59140166d3c41c47f858.png",
      "id": "bf4b1c1a29de76d555gik5",
      "ip": "1.1.1.1",
      "is_online": false,
      "lat": "53.24",
      "local_key": "eaaf39b2878f255d",
      "lon": "50.2",
      "model": "",
      "name": "1G Wifi switch(WIFI+BLE) 8",
      "product_id": "ven08ot7x16qxbty",
      "product_name": "一路开关（WIFI+BLE）",
      "sub": false,
      "time_zone": "+04:00",
      "update_time": 1740117761,
      "uuid": "8e063d12866694ba"
    }
  ],
  "success": true,
  "t": 1740198544661,
  "tid": "8c62a88bf0d511efa76bb217c6c149bc"
}
`
photo

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Malaysk, 2025-02-22:
After 3 days of trying, I managed to dump the firmware v.1.0.4

[readResult_BK7231T_QIO_BK7231T_2025-22-2-15-14-48.zip](https://github.com/user-attachments/files/18922145/readResult_BK7231T_QIO_BK7231T_2025-22-2-15-14-48.zip)

JSON
> {
	"rl1_lv":"1",
	"netled_lv":"1",
	"bt_type":"0",
	"bt1_pin":"7",
	"mxcl_led_m":"0",
	"rand_dpid":"18",
	"net_trig":"0",
	"led_dp":"15",
	"jv":"1.0.0",
	"backlit_select":"0",
	"nety_led":"0",
	"md":"0",
	"random":"0",
	"wfb64":"1",
	"stat":"0",
	"token":"null",
	"region":"null",
	"reg_key":"null",
	"dns_prio":"00nc_tp",
	"ssid":"null",
	"passwd":"null",
	"psk_key":"LRWRi9mpSX3ldfi2TNvrRWx5g0LeGJGGq0GHv",
	"auth_key":"wLhZxAvvAnW6owkhmTiDiFDs9jyfcceX",
	"ap_ssid":"SmartLife",
	"ap_passwd":"null",
	"country_code":"CN",
	"bt_mac":"null",
	"bt_hid":"null",
	"prod_test":"false }tled_pin",
	"total_stat":"2",
	"bt1_lv":"0",
	"reset_t":"5",
	"ch_flag1":"1",
	"inch_dp":"19",
	"module":"WB3S-IPEX",
	"dctrl_select":"0",
	"ch_cddpid1":"7",
	"init_conf":"14",
	"backlit_dp":"16",
	"cyc_dpid":"17",
	"ch_num":"1",
	"reuse_led_m":"0",
	"rl1_pin":"6",
	"netn_led":"0",
	"ch_dpid1":"1",
	"crc":"115"
}

Device configuration, as extracted from Tuya: 
- Button (channel 1) on P7
- Relay (channel 1) on P6
Device seems to be using WB3S-IPEX module, which is using BK7231T.
And the Tuya section starts, as usual, at 2023424

Cossid, 2025-02-22:
Added as `Tuya Generic Touch Switch v1.0.4`

closed 2025-02-22 by Cossid

Malaysk, 2025-02-23:
> Added as `Tuya Generic Touch Switch v1.0.4`

Hello! It works great! Thank you very much!
I did not open a new request, I will write here:
I have only one question:
I have exactly the same switch with firmware 1.0.4 only 2 gang - it is absolutely identical with 1 gang.
Is it necessary to make a full dump of the firmware, or is it possible to add this switch to the firmware database? I just had a hard time with the dump of 1 gang and I will need a lot of time to make 2 gang. Thanks in advance for your answer.

Cossid, 2025-02-23:
The same profile will work for exploiting via CloudCutter, but won't work for generating a config, as it won't have the same configuration data for the buttons, however you can use ESPHome Kickstart + LTChipTool UPK2ESPHome or OpenBeken tools to pull configuration after flashing.