tuya-cloudcutter/tuya-cloudcutter #790: Generic S18 IR Remote Control never gets past DHCP loop

fahhem, 2025-03-06:
It seems to be exploitable (even though batch number is 2024082401). However, I've run the script many, many times and dnsmasq never gets a DNS query, the device just keeps requesting a DHCP address endlessly.

```
Building cloudcutter docker image
[+] Building 0.5s (13/13) FINISHED                                                                                                                                             docker:default
 => [internal] load build definition from Dockerfile                                                                                                                                     0.0s
 => => transferring dockerfile: 462B                                                                                                                                                     0.0s
 => [internal] load .dockerignore                                                                                                                                                        0.0s
 => => transferring context: 2B                                                                                                                                                          0.0s
 => [internal] load metadata for docker.io/library/python:3.9.18-slim-bullseye                                                                                                           0.3s
 => [base 1/2] FROM docker.io/library/python:3.9.18-slim-bullseye@sha256:9ac27d4ecadc3ef02f980a8e2b37c7e8cdbf24039c5eddd9f98aff00d23b9e03                                                0.0s
 => [internal] load build context                                                                                                                                                        0.0s
 => => transferring context: 3.02kB                                                                                                                                                      0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto                     0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv                                                                                                                            0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                                                                                                                                      0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                                                                                                                                 0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy                                                                                             0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                                                                                                                               0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                                                                                                                                0.0s
 => exporting to image                                                                                                                                                                   0.0s
 => => exporting layers                                                                                                                                                                  0.0s
 => => writing image sha256:b191c84a62de2827047902339b7779cf83e2e169b5082bd4b0a411986703d02e                                                                                             0.0s
 => => naming to docker.io/library/cloudcutter                                                                                                                                           0.0s
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Occupied by systemd-resolve with PID 2793049.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Detected app armour
This has been known to block hostapd, which is required to complete the exploit
Do you wish to stop the app armour service? [y/N] 

Safety checks complete.

[?] Select your custom firmware file for BK7231N chip: ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
 ► ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
   OpenBeken-v1.18.31_bk7231n.ug.bin

Selected Device Slug: tuya-generic-s18-ir-repeater
Selected Profile: oem-bk7231n-irbox-mol-ty-2.0.0-sdk-2.3.1-40.00
Selected Firmware: ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP

Found access point name: "A-FE3C", trying to connect...
Device 'wlo2' successfully activated with 'deaf899b-0b4b-4d33-8bc5-998b393526de'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/UVwCVggLpccH.deviceconfig
[... 96 lines cut ...]
Mar  6 14:50:02 dnsmasq-dhcp[15]: 842012285 next server: 10.42.42.1
Mar  6 14:50:02 dnsmasq-dhcp[15]: 842012285 sent size:  1 option: 53 message-type  2
Mar  6 14:50:02 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 54 server-identifier  10.42.42.1
Mar  6 14:50:02 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 51 lease-time  12h
Mar  6 14:50:02 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 58 T1  6h
Mar  6 14:50:02 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 59 T2  10h30m
Mar  6 14:50:02 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  1 netmask  255.255.255.0
Mar  6 14:50:02 dnsmasq-dhcp[15]: 842012285 sent size:  4 option: 28 broadcast  10.42.42.255
Mar  6 14:50:02 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  3 router  10.42.42.1
Mar  6 14:50:02 dnsmasq-dhcp[15]: 842012285 sent size:  4 option:  6 dns-server  10.42.42.1
Mar  6 14:50:10 dnsmasq-dhcp[15]: 842012285 available DHCP range: 10.42.42.10 -- 10.42.42.40
Mar  6 14:50:10 dnsmasq-dhcp[15]: 842012285 client provides name: wlan0
...repeats...
```

I've tried the following:
* Followed the instructions at https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#my-device-gets-stuck-after-dhcp-what-can-i-do to no avail
* I run `tcpdump` and I see BOOTP/DHCP Request and Reply packets between the device and my computer.
* I've tried running dnsmasq on my host
* I've disabled firewalls and even cleared iptables to ensure nothing was dropping packets
* When I connect to the cloudcutterflash AP with my phone I see 1 DHCP back-and-forth, then lots of DNS queries as my phone is able to connect normally

Is there anything else I should be trying? Or is this device not actually exploitable?

Thank you!

Cossid, 2025-03-06:
MQTT is showing valetudo (open source smart vacuum firmware?) traffic.  Do you happen to have tailscale running on your network?  I believe it has a conflicting 10.42.x route

The device `wlo2` hitting dns is also abnormal, usually most tuya beken devices are named `wlan`

fahhem, 2025-03-06:
Thanks @Cossid! I had tailscale up, though it didn't use 10.42.x. I also had k3s up, which did use it in its CNI and flannel usage.

However, I've shut both down, even deleted the ip links so there's no reference to 10.42.x.x other than for wlo2, and it's still looping like before. And I also see the valetudo traffic (which is on my home network and was connecting to my mqtt server previously, but I shut it down)

Cossid, 2025-03-06:
I'd still retry the stuck FAQ method, and since it is making at least some connection, make sure distance is not an issue.  AP mode is not very strong.  Lastly, if your device has a button, try pressing (not holding) the button every few seconds, to help make sure it isn't trying to go into any power saving mode.

Beyond that, there isn't much more advice to give, aside from trying a more isolated environment.  Some devices are just finicky, which is why so many people also have issues with Tuya's platform.

fahhem, 2025-03-07:
Unfortunately, it's not getting stuck at that point, it just keeps repeating the DHCP request/offer and never moves on (nor does it go to sleep). The app approach doesn't get it to reconnect to `cloudcutterflash`, only retrying the script works for that.

Looks like this device isn't compatible after all, thanks anyway!

closed 2025-03-07 by fahhem