tuya-cloudcutter/tuya-cloudcutter #792: Device Dump: Door Sensor V1.0.3

DL5SMB, 2025-03-16:
Enclosed a dump (bin-file) of a Door Sensor which is not yet in the data base.
I used BK7231 Flasher, which could not find out the GPIO pinout. 

[readResult_BK7231N_QIO_window_door_2025-15-3-23-25-31.dmp](https://github.com/user-attachments/files/19275148/readResult_BK7231N_QIO_window_door_2025-15-3-23-25-31.dmp)

Cossid, 2025-03-16:
Added as `Tuya Generic Door Window Sensor v1.0.3`

There is no device configuration in storage, meaning you'll need to probe the pins manually.

closed 2025-03-16 by Cossid

DL5SMB, 2025-03-16:
[image]

[image]

[image]

I forgot the pictures. I will pinpoint it out. The sensor is U3, right?

Cossid, 2025-03-16:
Ah, this is a vibration sensor.  Yes, I'd assume U3 is the vibration sensor.