tuya-cloudcutter/tuya-cloudcutter #794: How to contribute: LEDVANCE E27 DIM Light 806lm - AC33908

Stunkymonkey, 2025-03-24:
Hey there,

I managed to flash `AC33908` with the following:

```bash
sudo ./tuya-cloudcutter.sh -p "ledvance-tunable-white-a60-b27-bulb-v1.1.4" -f ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin
```

now my question: How do i contribute this? So others can reuse this information as well.

I started: https://github.com/esphome/esphome-devices/pull/1061
And saw this file: https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/blob/master/devices/ledvance-tunable-white-a60-b27-bulb-v1.1.4.json (almost identical, but only with "white")

but I am not experienced enough with tuya-cloudcutter.

Cossid, 2025-03-24:
I mean the `storage json data` from the `source data` tab of UPK2ESPHome plugin within LTChipTool, and you'd need to re-install kickstart.  You can OTA back and forth between Kickstart and custom ESPHome.

Stunkymonkey, 2025-03-26:
i had first issues with api-key & ota-password. Now i removed all secret parts.

But flashing kickstart (`ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin`) via ota, i am getting:

```
Update Failed: Bad Magic Number
```

then i rebuild my firmware for openbeken and converted it via `ltchiptool` to uf2 to be able to flash it via ota. Yes i know now, that this was not smart...
the rest of my time today I now figured out to flash it back to esphome. So I was sadly going in a circle.

then i found: https://github.com/libretiny-eu/esphome-kickstart

but i am always getting "connection-reset" errors when trying to upload it via web-OTA or even update to my old image... seems like i am stuck?

Do you have a "better" way to flash esphome-kickstart?

Cossid, 2025-03-26:
You shouldn't be using ug.bin files if ESPHome is already installed, you should be using the .uf2 files.
You can brick devices using the wrong .ug.bin file, which is a part of the reason LibreTiny tries to avoid them.

https://docs.libretiny.eu/docs/flashing/esphome/#over-the-air-ota has instructions for switching back from OpenBeken to ESPHome.  But make sure you use the right chip type, or  you will brick the device leading to the requirement of a serial recovery.

Stunkymonkey, 2025-03-27:
and then ` ltchiptool plugin upk2esphome kickstart --storage-only` ?

<details><summary>results into</summary>

```
{
	"gw_bi": {
		"uuid": "SMp4cfQIbzGI",
		"psk_key": null,
		"auth_key": "vPA2VuLpTbyBFv7i",
		"ap_ssid": "A",
		"ap_passwd": null,
		"country_code": null,
		"bt_mac": null,
		"bt_hid": null,
		"prod_test": false
	},
	"user_param_key": {
		"Jsonver": "1.1.8",
		"brightmax": 100,
		"brightmin": 10,
		"c_lv": 1,
		"c_pin": 7,
		"cagt": 40,
		"category": "0501",
		"cmod": "c",
		"colormax": 100,
		"colormin": 10,
		"crc": 107,
		"cwmaxp": 100,
		"cwtype": 0,
		"defbright": 100,
		"defcolor": "c",
		"deftemp": 100,
		"dmod": 0,
		"module": "WB8P",
		"notdisturb": 0,
		"onoffmode": 1,
		"pmemory": 1,
		"prodagain": 0,
		"pwmhz": 600,
		"remdmode": 0,
		"rgbt": 0,
		"rstbr": 100,
		"rstcor": "c",
		"rstnum": 5,
		"rsttemp": 100,
		"title20": 1,
		"wfcfg": "prod",
		"wt": 0
	},
	"is_stride": 0,
	"gw_wsm": {
		"nc_tp": 4,
		"ssid": "Y2xvdWRjdXR0ZXJmbGFzaA==",
		"passwd": "YWJjZGFiY2Q=",
		"md": 0,
		"random": 0,
		"wfb64": 1,
		"stat": 2,
		"token": "AAAAAA",
		"region": "AA",
		"reg_key": null,
		"dns_prio": 2
	},
	"gw_di": {
		"abi": 0,
		"id": "909sT5X2lCi3lnOGjPEq",
		"swv": "1.1.4",
		"bv": "40.00",
		"pv": "2.2",
		"lpv": "3.3",
		"pk": "keyhxhxkrkkxcvt7",
		"firmk": "keyhxhxkrkkxcvt7",
		"cadv": "1.0.2",
		"cdv": "1.0.0",
		"dev_swv": "1.1.4",
		"s_id": "000001q4s9",
		"dtp": 0,
		"sync": 0,
		"attr_num": 0,
		"mst_tp_0": 0,
		"mst_ver_0": null,
		"mst_tp_1": 0,
		"mst_ver_1": null,
		"mst_tp_2": 0,
		"mst_ver_2": null,
		"mst_tp_3": 0,
		"mst_ver_3": null,
		"dminfo_name": null,
		"dminfo_code": null,
		"dminfo_report_code": null,
		"dminfo_sn": null
	},
	"gw_ai": {
		"key": "97yOZTulFnd3Rbqj",
		"lckey": "Knv0NFg1r1sQqZ0y",
		"h_url": "http://10.42.42.1/d.json",
		"h_ip": "10.42.42.1",
		"hs_url": null,
		"hs_ip": null,
		"hs_psk": "https://a3.tuyaeu.com/d.json",
		"hs_psk_ip": "18.158.227.228",
		"mqs_url": null,
		"mqs_ip": null,
		"mq_url": "10.42.42.1:1883",
		"mq_ip": "10.42.42.1",
		"ai_sp": null,
		"mq_psk": "m2.tuyaeu.com:8886",
		"mq_psk_ip": "3.121.210.75",
		"time_z": "+01:00",
		"s_time_z": "[]",
		"wx_app_id": null,
		"wx_uuid": null,
		"dy_tls_m": 1,
		"cloud_cap": 1025
	},
	"tls_ca_cnt": 0,
	"0000034p8t": [
		{
			"mode": "rw",
			"property": {
				"type": "bool"
			},
			"id": 20,
			"type": "obj"
		},
		{
			"mode": "rw",
			"property": {
				"range": [
					"white",
					"colour",
					"scene",
					"music"
				],
				"type": "enum"
			},
			"id": 21,
			"type": "obj"
		},
		{
			"mode": "rw",
			"property": {
				"min": 10,
				"max": 1000,
				"scale": 0,
				"step": 1,
				"type": "value"
			},
			"id": 22,
			"type": "obj"
		},
		{
			"mode": "rw",
			"property": {
				"type": "string",
				"maxlen": 255
			},
			"id": 25,
			"type": "obj"
		},
		{
			"mode": "rw",
			"property": {
				"min": 0,
				"max": 86400,
				"scale": 0,
				"step": 1,
				"type": "value"
			},
			"id": 26,
			"type": "obj"
		},
		{
			"mode": "wr",
			"property": {
				"type": "string",
				"maxlen": 255
			},
			"id": 28,
			"type": "obj"
		},
		{
			"mode": "rw",
			"id": 30,
			"type": "raw"
		},
		{
			"mode": "rw",
			"id": 31,
			"type": "raw"
		},
		{
			"mode": "rw",
			"id": 32,
			"type": "raw"
		},
		{
			"mode": "rw",
			"property": {
				"type": "bool"
			},
			"id": 101,
			"type": "obj"
		}
	],
	"000001q4s9": [
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"type": "bool"
			},
			"id": 20
		},
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"range": [
					"white",
					"colour",
					"scene",
					"music"
				],
				"type": "enum"
			},
			"id": 21
		},
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"min": 10,
				"max": 1000,
				"scale": 0,
				"step": 1,
				"type": "value"
			},
			"id": 22
		},
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"min": 0,
				"max": 1000,
				"scale": 0,
				"step": 1,
				"type": "value"
			},
			"id": 23
		},
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"type": "string",
				"maxlen": 255
			},
			"id": 25
		},
		{
			"type": "obj",
			"mode": "rw",
			"property": {
				"min": 0,
				"max": 86400,
				"scale": 0,
				"step": 1,
				"type": "value"
			},
			"id": 26
		},
		{
			"type": "obj",
			"mode": "wr",
			"property": {
				"type": "string",
				"maxlen": 255
			},
			"id": 28
		},
		{
			"mode": "rw",
			"id": 30,
			"type": "raw"
		},
		{
			"mode": "rw",
			"id": 31,
			"type": "raw"
		},
		{
			"mode": "rw",
			"id": 32,
			"type": "raw"
		},
		{
			"mode": "rw",
			"property": {
				"type": "bool"
			},
			"id": 101,
			"type": "obj"
		}
	],
	"em_sys_env": "BK7231S_2M",
	"mf_test_close": true
}
```

</details>

i then pasted it into https://upk.libretiny.eu/ and found the `c_pin` should be `9`. otherwise the light will not work.

Cossid, 2025-03-28:
Yes, that's the information I was looking for.

Really strange the pin configuration may be incorrect, I'm not sure how the device could have functioned with stock firmware if that is the case.

I have added a profile as `LEDVANCE SmartPlus AC33908 Dimmable Bulb v1.1.4` but left the P7 pin as it was, as we don't modify that data.

closed 2025-03-28 by Cossid

Stunkymonkey, 2025-03-28:
thanks for your help