tuya-cloudcutter/tuya-cloudcutter #795: Shell does not want to allow selection

treyus30, 2025-03-31:
[image]

On Rpi W2, I cannot select anything but the top value.

Cossid, 2025-04-01:
There is no reason this shouldn't work through SSH, that's the only way I've ever used it.

Alternately, you can try finding your profile at https://upk.libretiny.eu/ and copying the part after `?profile=` in the url and using the `-p` parameter with `-p <copied_profile_name>`

treyus30, 2025-04-03:
> There is no reason this shouldn't work through SSH, that's the only way I've ever used it.
> 
> Alternately, you can try finding your profile at https://upk.libretiny.eu/ and copying the part after `?profile=` in the url and using the `-p` parameter with `-p <copied_profile_name>`

Well that's certainly an interesting site. I ended up installing the docker container on an old Linux Mint laptop, but apparently my firmware is v1.3.10 for a generic Tuya 4 channel relay board, which doesn't exist. What next..?

treyus30, 2025-04-13:
I see this was closed.
However, the issue was not addressed, for the record. The fancy dynamic selection menu should be scrutinized for cross-platform compatibility.

Cossid, 2025-04-13:
Patches are welcome, but I can't reproduce this.  Others have successfully used Rpi Zero W2 over USB and via SSH as documented here: https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/HOST_SPECIFIC_INSTRUCTIONS.md#pi-zero-2w-with-ssh-over-usb
Incquirer carousel has always been the input method.  I can only assume it's something specific to your configuration, nobody else has reported being unable to scroll through the options.