tuya-cloudcutter/tuya-cloudcutter #797: Using RPi3b can't connect to device, always scanning, Treatlife SS01S and SK30

stricklerlj61, 2025-04-04:
I am just trying this and it never finds/connects to the device's AP even when I give the AP name. I can see the TreatLife AP with the RPi and my phone. I specified the AP using "sudo ./tuya-cloudcutter.sh -o TreatLife-SL-FA33" and it never connects. These are new items but they should be able to be connected to still, right? All I've heard is that they may not allow programming

Cossid, 2025-04-04:
Sounds like you probably don't have NetworkManager working correctly then.  Did you follow the instructions posted at https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/HOST_SPECIFIC_INSTRUCTIONS.md ?

stricklerlj61, 2025-04-04:
As far as I can tell I did all the steps, what do I do to verify I didn't screw it up?

Cossid, 2025-04-04:
How are you verifying your RPi can see the treatlife AP?  It must be able to detect it using `nmcli`, see the scripts for how it is scanning.

stricklerlj61, 2025-04-04:
I used "sudo iwlist wlan0 scan" and it shows up
I tried something from the scripts (nmcli device set ${WIFI_ADAPTER} managed yes) with and without sudo, putting wlan0 for wifi adapter and nmicli always says wlan0 is unavailable

Cossid, 2025-04-04:
Then your NetworkManager is not setup correctly, if it can't control wlan0 which matches what you'd use for iw

stricklerlj61, 2025-04-04:
I had to follow directions at
https://raspberrypi.stackexchange.com/questions/29783/how-to-setup-network-manager-on-raspbian/73816#73816
so it now shows up as disconnected, I'll try again

stricklerlj61, 2025-04-04:
The SS01S failed exploit but the SK30 succeeded

Thank you

closed 2025-04-04 by Cossid