tuya-cloudcutter/tuya-cloudcutter #799: Spectrum WOJ+14418 version v3.30

mkleijn, 2025-04-18:
Hi,

I tried flashing this CW/WW Filament bulb from Spectrum. I managed to flash another product from the same manufacturer (using a generic profile of the same SDK version), but this one has the exotic version v3.3.30 that I could not find in none of the profiles nor in the known patched devices list.

I tried the lighleak app to retrieve a dump, but I didn't manage to download anything (exploit succeeded, but error during dump extraction).

Has anyone seen this sdk version before? I don't really fancy cutting the device open and definitely don't trust my soldering skills, but as the device seems to be exploitable I guess it should be possible to flash OTA. Anyone any tips?

[image]

Cossid, 2025-04-20:
Firmware version 3.3.30 means it is an ESP8266 device, and not supported by CloudCutter.  3.3.30 is also patched against TuyaConvert.  Serial flashing will probably be your only option with these devices.

closed 2025-04-20 by Cossid

mkleijn, 2025-04-20:
Ok thanks for confirming @Cossid.