tuya-cloudcutter/tuya-cloudcutter #800: New Device - Amico 6" RGBCW Downlight

justlikeef, 2025-04-23:
I have been unable to get lightleak to successfully do a dump, but I was able to use cloudcutter to install 3rd party firmware.

Amico 6" RGBCW Downlight
https://www.amazon.com/dp/B0CQX7FP8K?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1
Model# DC08-DR6-12W120-CCT+W5

reflashed using: oem-bk7231n-ceiling-light-ty-1.2.8-sdk-2.3.1-40.00.json

ESPHOME YAML:

esphome:
  name: amico-downlight
  friendly_name: amico-downlight

bk72xx:
  board: cbu

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "[key]"

ota:
  - platform: esphome
    password: "[password]"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Amico-Downlight"
    password: "[password]"

captive_portal:

web_server:

mdns:

button:
  - platform: restart
    name: Restart

debug:
  update_interval: 30s

text_sensor:
  - platform: debug
    reset_reason:
      name: Reset Reason
  - platform: libretiny
    version:
      name: LibreTiny Version

sensor:
  - platform: uptime
    name: Uptime

output:
  - platform: libretiny_pwm
    id: output_red
    pin: P26
  - platform: libretiny_pwm
    id: output_green
    pin: P24
  - platform: libretiny_pwm
    id: output_blue
    pin: P6
  - platform: libretiny_pwm
    id: output_warm
    pin: P8
  - platform: libretiny_pwm
    id: output_cold
    pin: P7

light:
  - platform: rgbww
    id: light_rgbww
    name: Light
    color_interlock: true
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    red: output_red
    green: output_green
    blue: output_blue
    cold_white: output_cold
    warm_white: output_warm

Cossid, 2025-04-23:
Without at least a storage dump, there is no action to be taken other than marking this as documentation for others who may have the same device.

closed 2025-04-23 by Cossid

justlikeef, 2025-04-23:
Is there a way I can provide the dump other than with lightleak?
Thanks,
Rob

Cossid, 2025-04-23:
If you install Kickstart, you can save the storage data with LTChipTool's UPK2ESPHome plugin and attach it here.

Cossid, 2025-04-25:
Added as `Amico DC08-DR6-12W120-CCT+W5 6in RGBCT Downlight v1.2.8`