tuya-cloudcutter/tuya-cloudcutter #802: Flashing Wipro smart batten 20W (Tuya backend) using tuya convert tool fails

ShunMaster999, 2025-04-25:
I am trying to flash the firmware for Wipro smart batten 20W (Tuya backend) however upon running the script I encounter the following - 

./start_flash.sh 
tuya-convert v2.4.6
Checking for network interface wlan1... Found.
Checking UDP port 53... Occupied by dnsmasq with PID 813.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] y
Attempting to stop dnsmasq.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 797.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to stop mosquitto.service
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen.
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
....................................
SmartConfig complete.
Resending SmartConfig Packets
.....................................................
SmartConfig complete.
Resending SmartConfig Packets
.....................................................
Timed out while waiting for the device to (re)connect
Stopping smart config
======================================================
Attempting to diagnose the issue...
An ESP82xx based device connected according to your wifi log.
If this is the device you are trying to flash, another issue may be preventing it from flashing.
Otherwise, it could be that the device does not use an ESP82xx or it did not connect.
For additional information, check the *.log files inside the scripts folder.
Please include these logs when opening a new issue on our GitHub issue tracker.
======================================================
Do you want to try flashing another device? [y/N] n
======================================================
Cleaning up...
No screen session found.
No screen session found.
Closing AP
Exiting...

I have attached the logs -

[smarthack-mqtt.log](https://github.com/user-attachments/files/19912028/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/user-attachments/files/19912027/smarthack-psk.log)
[smarthack-udp.log](https://github.com/user-attachments/files/19912025/smarthack-udp.log)
[smarthack-web.log](https://github.com/user-attachments/files/19912024/smarthack-web.log)
[smarthack-wifi.log](https://github.com/user-attachments/files/19912026/smarthack-wifi.log)

Cossid, 2025-04-25:
This is the Tuya-CloudCutter issues section, but you've included Tuya-Convert logs.  The logs indicate the device you have was likely an ESP8266 device (but that may not be entirely accurate, but I think they do mac address comparisons), but could not be exploited with Tuya-Convert.  Tuya-CloudCutter will not work with ESP8266 devices.  Your only option will likely be to serial flash this device.

closed 2025-04-25 by Cossid

Cossid, 2025-04-28:
That is outside the scope of this project.