tuya-cloudcutter/tuya-cloudcutter #808: I want to submit a dump - Not sure how

Deep-Jiwan, 2025-05-30:
My apologies if this is the wrong place to post this....

I have the storage data JSON of a Tuya generic LSPA7 Wi-Fi socket plug.

There exists a profile on upk.libretiny.eu , but it didn't work for me fully. ( only the relay worked ).

I wish to submit a ltchiptool storage data and other dump files needed to add like a v2 for the generic lspa7 plug. 

My current config has all the telemetry and functionalities working except a blue led indicator for wifi connectivity.
+
If you would like this json or other data please let me know how I can contribute it without revealing any sensitive information. 

Thank you..

Wifi plug reference : eWelink UK Plug 1PCs
https://a.aliexpress.com/_EQmNRW2

Cossid, 2025-06-01:
You can attach the `source data` from LTChipTool's UPK extension, and we can see.

Deep-Jiwan, 2025-06-06:
{
    "gw_bi": {
        "uuid": "eYCwjM9tqmcv",
        "psk_key": null,
        "auth_key": "uYPGwG4txR3OcAEq",
        "ap_ssid": "A",
        "ap_passwd": null,
        "country_code": null,
        "bt_mac": null,
        "bt_hid": null,
        "prod_test": false,
        "fac_pin": null
    },
    "user_param_key": {
        "bt1_lv": 0,
        "bt1_pin": 10,
        "bt1_type": 0,
        "ch1_stat": 2,
        "ch_cddpid1": 9,
        "ch_dpid1": 1,
        "ch_num": 1,
        "chip_type": 0,
        "crc": 39,
        "ele_fun_en": 1,
        "ele_pin": 7,
        "ffc_select": 0,
        "jv": "1.0.5",
        "lose_vol": 70,
        "module": "CB2S",
        "net_trig": 2,
        "netled1_lv": 0,
        "netled1_pin": 8,
        "netled_reuse": 0,
        "netn_led": 0,
        "nety_led": 1,
        "over_cur": 16000,
        "over_vol": 300,
        "reset_t": 5,
        "resistor": 1,
        "rl1_lv": 1,
        "rl1_pin": 26,
        "rl1_type": 0,
        "sel_pin_lv": 1,
        "sel_pin_pin": 24,
        "vi_pin": 6,
        "vol_def": 0
    },
    "gw_di": {
        "abi": 0,
        "id": "4eHlc4SH2YdAiL9mckYv",
        "swv": "1.1.8",
        "bv": "40.00",
        "pv": "2.2",
        "lpv": "3.3",
        "pk": "keyjup78v54myhan",
        "firmk": "keyjup78v54myhan",
        "cadv": "1.0.3",
        "cdv": "1.0.0",
        "dev_swv": "1.1.8",
        "s_id": "0000038vp3",
        "dtp": 0,
        "sync": 0,
        "attr_num": 0,
        "mst_tp_0": 0,
        "mst_ver_0": null,
        "mst_tp_1": 0,
        "mst_ver_1": null,
        "mst_tp_2": 0,
        "mst_ver_2": null,
        "mst_tp_3": 0,
[... 216 lines cut ...]
            },
            "id": 43
        }
    ],
    "is_stride": 0,
    "timer_arr": {
        "lastFetchTime": 0,
        "cnt": 0
    },
    "em_sys_env": "bk7231n"
}

W: PyInstaller\loader\pyimod02_importers.py:450: UserWarning: ltchiptool.gui.base.zc is deprecated, migrate to ltchiptool.gui.mixin.zc

I: Found new device: COM4 - Standard Serial over Bluetooth link
I: Found new device: COM5 - Standard Serial over Bluetooth link
I: UPK: Found BK7231N config!
I: UPK: Switch/plug config
I: UPK:  - relay 1: pin P26
I: UPK:  - button 1: pin P10
I: UPK: Status LED: pin P8, inverted True
I: UPK: Power monitoring chip BL0937: CF/ELE=P7, CF1/VI=P6, SEL=P24
I: UPK:  - shunt resistor: 1 mΩ

Sorry for the delay

closed 2025-06-06 by Cossid