tuya-cloudcutter/tuya-cloudcutter #809: 404 GET loop

16BitDoggo, 2025-06-02:
My device is a Sylvania 75702

I tried to give it ESPHome fw, but I see something like this:
`404 GET /v2/app management/web/appgrid (ip in parentheses) 1.47 ms
404 GET /v2/app management/web/appgrid (ip in parentheses) 1.47 ms
404 GET /v2/app management/web/appgrid (ip in parentheses) 1.47 ms
404 GET /v2/app management/web/appgrid (ip in parentheses) 1.47 ms
404 GET /v2/app management/web/appgrid (ip in parentheses) 1.47 ms
404 GET /v2/app management/web/appgrid (ip in parentheses) 1.47 ms
404 GET /v2/app management/web/appgrid (ip in parentheses) 1.47 ms
404 GET /v2/app management/web/appgrid (ip in parentheses) 1.47 ms
404 GET /v2/app management/web/appgrid (ip in parentheses) 1.47 ms
404 GET /v2/app management/web/appgrid (ip in parentheses) 1.47 ms
404 GET /v2/app management/web/appgrid (ip in parentheses) 1.47 ms
404 GET /v2/app management/web/appgrid (ip in parentheses) 1.47 ms
404 GET /v2/message_bus /socket.io/?EIO=3&transport=websocket 404 GET (ip in parentheses) 1.47 ms`

it has looped almost exactly like this (ping obviously fluctuates) for around an hour and shows no signs of stopping.

can anyone help?

Cossid, 2025-06-02:
What is the IP, and what DHCP request does it correspond to (may need to run with `-v`)?  That looks like a mobile device connected to the CloudCutter AP making requests.  That isn't a request we've ever seen a Tuya device make.

A full log would be more helpful.

Cossid, 2025-06-02:
That's not even the correct range, so you have something forcing routes.  CloudCutter's AP runs the 10.42.42.0/24 range.
If you have TailScale or some other VPN, you'll need to disable it.

Cossid, 2025-06-02:
I can't speak to that.  Either way, it's not the Tuya device making requests, so you have issues we can't really support here.

Again, full logs would be more helpful.

16BitDoggo, 2025-06-02:
not 100% sure how to dump.

is ctrl+c okay or will it brick the plug?

16BitDoggo, 2025-06-02:
im sending this from a windows pc, the server pc is running debian with gui. any way to work around that?

there is a disney circle on my network... could that affect things?

Cossid, 2025-06-02:
You just need to copy/paste the full output from the ssh session/terminal after you start the CloudCutter script.  Selecting the text should copy it to your clipboard and you should be able to paste it.

I have no idea what a disney circle is, I cannot comment on that.  It looks like some sort of filtering device, so yes, it's possible it is causing issue.

16BitDoggo, 2025-06-03:
Curiously, I looked on my network to find that IP is my Windows PC (the one I'm writing this from). I don't think I have any programs that have any reason to scan the network. Any clue?

Cossid, 2025-06-03:
No idea, but either way, 404 is the correct response, as it's unrelated to CloudCutter.  Again, that isn't a Tuya device making that request.  Generally, you can ignore all requests made by non-tuya devices, it will always respond 404 to them.

16BitDoggo, 2025-06-03:
while i was trying to get the logs, the plug is stuck with the led on constantly