tuya-cloudcutter/tuya-cloudcutter #810: IR Universal Remote Controller w/T&H sensor v1.0.2 - no 3d party firmware flash possible

rocord01, 2025-06-06:
https://www.elektroda.com/rtvforum/topic3964328.html

[image]

```
sudo ./tuya-cloudcutter.sh
Building cloudcutter docker image
[+] Building 0.3s (13/13) FINISHED                                                                                          docker:default
 => [internal] load build definition from Dockerfile                                                                                  0.0s
 => => transferring dockerfile: 465B                                                                                                  0.0s
 => [internal] load metadata for docker.io/library/python:3.9.18-slim-bullseye                                                        0.2s
 => [internal] load .dockerignore                                                                                                     0.0s
 => => transferring context: 2B                                                                                                       0.0s
 => [base 1/2] FROM docker.io/library/python:3.9.18-slim-bullseye@sha256:9ac27d4ecadc3ef02f980a8e2b37c7e8cdbf24039c5eddd9f98aff00d23  0.0s
 => [internal] load build context                                                                                                     0.0s
 => => transferring context: 3.02kB                                                                                                   0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essenti  0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv                                                                         0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                                                                                   0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                                                                              0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy                                          0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                                                                            0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                                                                             0.0s
 => exporting to image                                                                                                                0.0s
 => => exporting layers                                                                                                               0.0s
 => => writing image sha256:b29eba224e6d13a2dce591162b0301739731043127d0b22ea667365febef3bb3                                          0.0s
 => => naming to docker.io/library/cloudcutter                                                                                        0.0s
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 ► By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 1.0.2 - BK7231N / bk7231n_common_user_config_ty
   1.0.18 - BK7231N / oem_bk7231n_slide_strip_ty
   1.0.2 - BK7231N / 19DWT_CBU_7231NL_1
   1.0.2 - BK7231N / 812WT_PIR
   1.0.2 - BK7231N / 898WT_CBU_7231NL
   1.0.2 - BK7231N / LGB88
   1.0.2 - BK7231N / QC0294_P1_QMCBU_CEILING5
 ► 1.0.2 - BK7231N / bk7231n_common_user_config_ty
   1.0.2 - BK7231N / dy_co_400a_cbu
   1.0.2 - BK7231N / fj_wifi_ble_controller_moto
   1.0.2 - BK7231N / oem_bk7231n_cutout_self_switch
   1.0.2 - BK7231N / oem_bk7231n_dctrl_switch
   1.0.2 - BK7231N / oem_bk7231n_light5_hailai
   1.0.2 - BK7231N / oem_bk7231n_safe_dltj_plug

Performing safety checks to make sure all required ports are available
Checking UDP port 53... Occupied by systemd-resolve with PID 950118.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Detected app armour
This has been known to block hostapd, which is required to complete the exploit
Do you wish to stop the app armour service? [y/N] y
Unloading AppArmor profiles
/lib/apparmor/apparmor.systemd: 273: printf: printf: I/O error
AppArmour has been turned off.  You will need to manually restart it or reboot your OS for it to turn back on.
Safety checks complete.

[?] Select your custom firmware file for BK7231N chip: OpenBeken-v1.18.31_bk7231n.ug.bin
   ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin
 ► OpenBeken-v1.18.31_bk7231n.ug.bin

Selected Device Slug: eyesound-y13-smart-bedside-lamp
Selected Profile: bk7231n-common-user-config-ty-1.0.2-sdk-2.3.1-40.00
Selected Firmware: OpenBeken-v1.18.31_bk7231n.ug.bin

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
...
Found access point name: "SmartLife-E720", trying to connect...
Device 'wlx98254a2f4679' successfully activated with 'c0378d82-b951-4b6b-ba7d-c5eea281c0cc'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/YegtvjHTskkh.deviceconfig
Saved device config in /work/configured-devices/YegtvjHTskkh.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
...
Found access point name: "SmartLife-E720", trying to connect...
Device 'wlx98254a2f4679' successfully activated with 'c0378d82-b951-4b6b-ba7d-c5eea281c0cc'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================
```

This device is also sometimes referred to as an "Aubess IR Remote Controller", but tuya-cloudcutter only seems to have a firmware 2.0.0 profile for that device.

Cossid, 2025-06-06:
That just means it doesn't have a TuyaMCU firmware, which is mostly expected.  Most IR blasters use the `irbox-mol-ty` firmware, of which, there is not a profile for 1.0.2 yet, so a new dump would be needed to support it (though it could have a different class as well).

Cossid, 2025-07-15:
Closing as stale/unsupported.  Feel free to re-open if you are able to provide a dump.

closed 2025-07-15 by Cossid