tuya-cloudcutter/tuya-cloudcutter #811: SmartSocket TP22 will not be flashed

domerich, 2025-06-08:
Hi people I am new to cloudcutter and now I am stuck

I am running Fedora 42 on my Macbook Pro M1.
I have a couple of SmartSocket TP22 with Tuya on them lying around so I though I would give it a try. Not sure what chip they have inside and I dont want to break them open.

So I tried to run

fedora:/var/lib/mock/fedora-42-aarch64/result/tuya-cloudcutter$ sudo ./tuya-cloudcutter.sh -s mywifi mywifipassword -r

Because I dont know what to choose I chose

BY firmware

 ► 1.1.8 - BK7231N / oem_bk7231n_plug

 ► Tuya Generic

 ► LSPA5BP 10A Breaker

and it will go like this

`erforming safety checks to make sure all required ports are available
Checking UDP port 53... Occupied by systemd-resolve with PID 97840.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service

Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Attempting to stop firewalld.service
When done, enable with: sudo systemctl start firewalld.service
Safety checks complete.

Selected Device Slug: tuya-generic-lspa5bp-10a-breaker
Selected Profile: oem-bk7231n-plug-1.1.8-sdk-2.3.1-40.00

Found access point name: "SmartLife-0A8C", trying to connect...
Gerät »wlp1s0f0« wurde erfolgreich mit »c0c2881c-9526-447b-b1f0-3a2c8c913ac9« aktiviert.
Error, could not connect to SSID.
Failed to connect, please run this script again

`

any ideas where I could have gone wrong? Is this device simply not supported?

Cossid, 2025-06-08:
Based on that log, I'd guess Network Manager is not fully installed/configured, which is a requirement for CloudCutter.

Additionally, you can follow https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out to help narrow down which chipset is inside your device.

Cossid, 2025-06-30:
Closing as stale, as this sounds like a system configuration issue, not an issue with CloudCutter.

closed 2025-06-30 by Cossid