tuya-cloudcutter/tuya-cloudcutter #813: Tuya Generic Energy Meter

toreupfeet, 2025-06-13:
I've tried many times to flash my Tuya Generic Energy Meter with ESPHome. The device pairs, the config is apparently saved, and when I power cycle and pair again the end result is always unsuccessful. Perhaps this is because the Tuya firmware on the device shows v1.0.13 and the current Cloud Cutter solution appears to be made for v1.0.10.

Any advice would be welcome.

Cossid, 2025-06-13:
Correct, if the firmware version doesn't match, it will never work.

However, it does look like the v1.0.10 profile offered and update to v1.0.13 and it is also not patched yet, so I have added a profile for v1.0.13 you may try.

closed 2025-06-13 by Cossid

toreupfeet, 2025-06-13:
Thank you for the reply. I really appreciate it. I tried the v1.0.13 profile a moment ago, twice, and it did not work. Just an FYI.

I am using Linux Mint, and during the Cloud Cutter process I am prompted to stop `systemd-resolve` and `apparmor`, which I always stop successfully. Then the device pairs, saves the config, I am prompted to power cycle, config is wiped, device pairs, and then the process is ultimately not successful.

I have not used Tuya CC before so I wanted to describe the details in case maybe I am doing something wrong.

toreupfeet, 2025-06-14:
Thank you again. I had no luck with either of the v1.0.13 profiles. I really appreciate the reply.

toreupfeet, 2025-06-19:
Question: Is there any chance of a fix for the 80A meter with v1.0.13 in the future? If not, I'll return the meter while I still can.