tuya-cloudcutter/tuya-cloudcutter #823: Dump of UltraPro WFD1001 USA outlet socket

eyal0, 2025-07-06:
I took a dump of this [UltraPro WFD1001 ](https://www.amazon.com/UltraPro-Receptacle-Compatible-Tamper-Resistant-51417/dp/B0BRDGHTBD)USA outlet socket.  Dump is attached.

[UltraPro-WFD1001-Wall-Outlet.zip](https://github.com/user-attachments/files/21090567/UltraPro-WFD1001-Wall-Outlet.zip)

[image]

[image]

[image]

It says that it cannot have a yaml automatically generated for it because it uses TuyaMCU or something like that.  The WFD4001 product says that same thing.  Nonetheless, I have had success with using `tuya-cloudcutter` to OTA flash the WFD4001.  So I hope that I will also be able to OTA flash this product, the WFD1001, too.

Cossid, 2025-07-06:
Added as `UltraPro WFD1001 Wall Outlet v1.0.4`

Pin data wasn't in storage, but rather store in the app.  It has been extracted and is available at https://upk.libretiny.eu/?profile=ultrapro-wfd1001-wall-outlet-v1.0.4

closed 2025-07-06 by Cossid

eyal0, 2025-07-06:
Oh wow, so it can be OTA flashed?  This is great for me!  I did not enjoy the soldering!  Thanks for your help!