tuya-cloudcutter/tuya-cloudcutter #824: Dump of Topeler YSW102 Smart Wall Outlet with 2 Individually Controlled Ports

eyal0, 2025-07-07:
It has two LEDs, one per outlet, and a third LED for config purposes, I suppose.

https://www.amazon.com/dp/B0C741MSY2
The chip is CB2S.
The Firmware on Smart Life claims to be main module version 1.1.15 with "Other type" v1.0.2.
Maybe it's the same as this: https://www.elektroda.com/rtvforum/topic4046153.html but mine seems to have a different version of firmware on it.

[topeller-two-relay-outlet.zip](https://github.com/user-attachments/files/21094817/topeller-two-relay-outlet.zip)

[image]
[image]
[image]
[image]

Cossid, 2025-07-07:
Plug firmware 1.1.15 is already on the [known patched firmware list](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware), serial flashing will remain the only option for this device.

closed 2025-07-07 by Cossid

eyal0, 2025-07-07:
Bummer! Thanks!