tuya-cloudcutter/tuya-cloudcutter #825: Profile Request: Cecotec Ready Warm 6750 Crystal Connection (PID: n72zzd550zyw8qte, FW: 2.6.22)

adrianipopescu, 2025-07-12:
```
device_name: Cecotec Ready Warm 6750 Crystal Connection
product_id: n72zzd550zyw8qte
firmware_version: 2.6.22
firmware_key: keyjkajwawthq4f5
uuid: afa4e9a2a5a6b2d1
mac_vendor: Tuya Smart Inc.
ip_address: 10.0.100.87
category: qn
model: 5318/05319/05320/05350/05351/05352

expected_chipset: BK7231N or BK7231T (likely using WB3S or CB2S module)
scan_result: Port 6668 open, all others closed (no HTTP/MQTT UI)
nmap_output: |
  PORT     STATE SERVICE VERSION
  6668/tcp open  irc?

device_access: Wall-mounted; unable to open for now to visually confirm module

notes: >
  Requesting CloudCutter profile support for this Tuya-based heater. 
  Appears to use a BK7231 module based on MAC vendor, firmware, and nmap scan. 
  No current OTA profile exists for firmware 2.6.22 / keyjkajwawthq4f5.
  If a firmware dump is needed, I may be able to open another identical unit in the future.
```

Cossid, 2025-07-12:
Firmware 2.6.22 is for a TuyaMCU device with a Realtek RTL8720CF module.  There is no CloudCutter support for Realtek at this time.

adrianipopescu, 2025-07-12:
@szupi-ipuzs yes please

szupi-ipuzs, 2025-07-12:
@adrianipopescu 

Here's the yaml (partial)

```
esp8266:
  board: esp_wroom_02
  restore_from_flash: True

uart:
  - id: tuya_uart
    rx_pin: GPIO13
    tx_pin: GPIO15
    baud_rate: 9600

tuya:

climate:
  - platform: "tuya"
    name: "Heater"
    switch_datapoint: 1
    target_temperature_datapoint: 3
    current_temperature_datapoint: 4
    supports_heat: True
    supports_cool: False
    visual:
      min_temperature: 15
      max_temperature: 35
      temperature_step: 1

switch:
  - platform: "tuya"
    name: "Lock"
    switch_datapoint: 2

number:
  - platform: "tuya"
    name: "Timer"
    number_datapoint: 5
    min_value: 0
    max_value: 1440
    step: 1
    icon: "mdi:clock"
    unit_of_measurement: "min"

select:
  - platform: "tuya"
    name: "Mode"
    enum_datapoint: 7
    optimistic: True
    options:
      0: Boost
      1: Eco
```

Still @Cossid might be right about difficulty with flashing your device. I've bought mine second-hand that is why it might still have an older hardware with tywe1s. The different firmware version and product id seem to suggest so too.