tuya-cloudcutter/tuya-cloudcutter #841: Tuya-cloudcutter: Temperature Humidity Wifi Tuya SHP-LB3

yennhitbs1990-ship-it, 2025-09-16:
I need to flash the temperature and humidity sensor of the Wifi server Tuya SHP-LB3. However, there is currently no profile for this device.

Device link.

https://tuya.vn/cam-bien-giam-sat-nhiet-do-do-am-phong-may-chu-wifi-tuya-shplb3-p31903648.html

Can any experts help me? Thank you very much!

yennhitbs1990-ship-it, 2025-09-16:
If you can, please give me profile of Tuya SHP-LB3v2 and Tuya SHP-LB3v3 devices.

https://tuya.vn/cam-bien-giam-sat-nhiet-do-do-am-phong-may-chu-wifi-tuya-shplb3v2-p34021813.html

https://tuya.vn/cam-bien-giam-sat-nhiet-do-do-am-phong-may-chu-man-hinh-lon-wifi-tuya-shplb3v3-p34136052.html

Thanks you very very very ... much !

Cossid, 2025-09-16:
We cannot add a profile without a full device dump to support the device, assuming it's not already patched.  Many battery/TuyaMCU devices are already patched, so I'd start by comparing the version against the [known patched list](https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware), and if it's not present there, try the by firmware version/name method, and if there is no match, a [full firmware dump](https://github.com/tuya-cloudcutter/tuya-cloudcutter#device-dumps) would be required.

If you are able to provide a dump, we can re-open the issue and add a profile, provided it is not already patched.

closed 2025-09-16 by Cossid

yennhitbs1990-ship-it, 2025-09-18:
Dear @Cossid ,

I know, this is not a service. It is your contribution and the contribution of the Github community for a better future. I am very grateful for that.

My device is using firmware version V1.0.0 as shown below.

The device does not need to use battery.

Thank you very much again!

[image]

Cossid, 2025-09-18:
I'd suggest trying the `Tuya-Generic_SNT957W---TDE-Temperature-Sensor-with-Probe-v1.0.0` profile.  If that doesn't work, you would need to supply a dump to be analyzed.